espressif/esp-lwip #13: Assertion failed: sockets[i].select_waiting == 0

pschlang, 2020-02-19:
In situations where the max number of sockets is close to the number of actually used sockets, the chance of hitting a race condition in lwIP increases which causes a system reset due to an assertion:

`Assertion failed: sockets[i].select_waiting == 0`

This has been fixed upstream in https://savannah.nongnu.org/bugs/?57445 / http://git.savannah.nongnu.org/cgit/lwip.git/commit/?id=8d7e436a9d24a2a1ee35aac72a7055e80330e994 and it would be useful to merge it here as well.

Alvin1Zhang, 2020-04-02:
@pschlang Thanks for reporting.

freakyxue, 2020-04-02:
hi @pschlang 
I've met this problem before. I think PR is a better solution

freakyxue, 2020-04-02:
hi @pschlang 
I think there's another solution。
You can add this patch for testing.
dd3b301750f19c9f15248ac32d366cd41c6b3225

pschlang, 2020-04-02:
Thanks @freakyxue, but I think it is better to merge the upstream patch as it will solve the root cause and not just hide the symptom.

MightyPork, 2020-04-13:
Is there any work-around we can use before this is merged/fixed?

The use case is simple and I would dare say common:
1. select waiting on a socket
2. in another thread, the socket FD is closed
3. lwip assert

pschlang, 2020-06-22:
This is very easy to patch, I don't really understand why there's no activity here from ESP side.

We currently have to patch this manually every time we update the esp-idf SDK.

freakyxue, 2020-06-28:
hi @pschlang 
I'm sorry I didn't reply in time,
This patch has been added to IDF, and we will merge it as soon as possible

closed 2020-08-03 by espressif-bot