espressif/esp-lwip #17: build error, like .h file not include?

betashepherd, 2020-07-15:
from /opt/Espressif/esp-idf/components/esp_event/default_event_loop.c:15:
/opt/Espressif/esp-idf/components/lwip/lwip/src/include/lwip/ip.h:78:3: error: unknown type name 'ip_addr_t'
   ip_addr_t local_ip; \
   ^
/opt/Espressif/esp-idf/components/lwip/lwip/src/include/lwip/ip.h:91:3: note: in expansion of macro 'IP_PCB'
   IP_PCB;
   ^
/opt/Espressif/esp-idf/components/lwip/lwip/src/include/lwip/ip.h:79:3: error: unknown type name 'ip_addr_t'
   ip_addr_t remote_ip; \
   ^
/opt/Espressif/esp-idf/components/lwip/lwip/src/include/lwip/ip.h:91:3: note: in expansion of macro 'IP_PCB'
   IP_PCB;

Alvin1Zhang, 2020-07-24:
@betashepherd Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

david-cermak, 2020-08-20:
@betashepherd Please reopen with more details on reproducing your issue. Closing for now. If you're experiencing issues with building IDF code (`default_event_loop.c`), please prefer raising a new issue in IDF repo.

closed 2020-08-20 by david-cermak