espressif/esp-lwip #28: lwip_getaddrinfo: EAI_SERVICE for port 0

boborjan2, 2021-01-06:
Hi,

lwip_getaddrinfo() incorrectly returns error EAI_SERVICE for a presumably valid call: (1 it runs on major ip stacks like the one of Linux and Windows, 2)  it should be fine according to the manual of getaddrinfo)
```
struct addrinfo hints, *res = NULL;
char serv[6] = "0";

memset(&hints, 0, sizeof(hints));
/* set-up hints structure */
hints.ai_family   = af;
hints.ai_flags    = AI_PASSIVE | AI_NUMERICHOST;
hints.ai_socktype = SOCK_DGRAM;
hints.ai_protocol = IPPROTO_UDP;
error = getaddrinfo(NULL, serv, &hints, &res);   --> EAI_SERVICE
```
Debugging shows that the return is performed in netdb.c, near line 305, in function lwip_getaddr():
```
if (servname != NULL) {
    /* service name specified: convert to port number
     * @todo?: currently, only ASCII integers (port numbers) are supported (AI_NUMERICSERV)! */
    port_nr = atoi(servname);
    if ((port_nr <= 0) || (port_nr > 0xffff)) {
      return EAI_SERVICE;
    }
 }
```
To my understanding. port_nr == 0 is a valid value for dynamic port allocation, and bind() will handle it accordingly. A possible solution would be:
 ```
if ((port_nr < 0) || (port_nr > 0xffff)) {
      return EAI_SERVICE;
 }
```

Thanks,
Viktor

Alvin1Zhang, 2021-01-07:
Thanks for reporting.

yuanjianmin, 2021-01-22:
hi @boborjan2 , thanks for reporting this issue. I also test your code on Linux platform. The return value of  getaddrinfo is 0 which is the same as you expected. Maybe this is a native bug of lwip. You can report this bug to [upstream](http://savannah.nongnu.org/bugs/?group=lwip). We will fix the problem synchronously.  Thank you.

closed 2021-01-27 by igrr