espressif/esp-lwip #30: The dhcp client behaviour is not standard compliant for DHCP clients (RFC2131)

mitec-embedded, 2021-04-21:
Starting from commit 2117f049dca64d15a12e81a308c27f36ac6e6272 which adds option 61 (Client identifiers) the dhcp is no more complaining for standard RFC2131, since the standard says 
`If the client
   supplies a 'client identifier', the client MUST use the same 'client
   identifier' in all subsequent messages, and the server MUST use that
   identifier to identify the client.`

The option 61 was only included in "discover" and "select" messages, but not in the dhcp_renew() or other methods.
This is causing a big issue:

When the esp32 first boot, it sends both the DISCOVER and DHCP_REQUEST containing the option "client id".
But then, after a while, when the ESP try to renew its IP, it sends a DHCP_REQUEST messages WITHOUT the "client id".
This is causing that "professional DHCPs" like zyxell firewalls that slavishly follow the DHCP standard, sends a NACK.

This cause the ESP stack to drop all connections and restart a DHCP negotiation. And this cycles over forever, since some requests has the client_id, and some others has not.

david-cermak, 2021-04-22:
@mitec-embedded Thanks for reporting this bug!

Indeed this option should be added to other packets as well. This needs to be fixed. Please find a quick patch below:
[dhcp-Fix-client-identifier-to-be-added-to-relevant-p.patch.txt](https://github.com/espressif/esp-lwip/files/6358310/dhcp-Fix-client-identifier-to-be-added-to-relevant-p.patch.txt)

closed 2021-05-07 by igrr

1e1, 2021-07-24:
Dear @david-cermak, 
Do you have any news about a possible PR?