espressif/esp-lwip #36: To sync with upstream lwIP-v2.1.3 release

AxelLin, 2021-11-11:
lwIP 2.1.3 is now available which is a bugfix-only release.
All users are strongly suggested to upgrade!

https://savannah.nongnu.org/forum/forum.php?forum_id=10072

david-cermak, 2021-12-15:
Thanks for posting this issue, will keep it open to track down the progress on upgrading to `2.1.3`

AxelLin, 2022-02-10:
Please consider this as bug fix since lwIP 2.1.3 is a bugfix-only release.

BTW, it's quite misleading that esp-lwip version is 2.1.3 now. It makes people think it's already based on upstream 2.1.3.
https://github.com/espressif/esp-lwip/blob/2.1.2-esp/src/include/lwip/init.h#L53-L57

AxelLin, 2022-08-22:
@david-cermak 
Just to confirm if this is done by 2.1.3-esp? If so, you can close this issue.
Thanks.

david-cermak, 2022-08-22:
@AxelLin Thanks for the reminder. Yes, this could be closed.

Migration to `2.1.3` has been completed and current IDF master references [`2.1.3-esp`](https://github.com/espressif/esp-lwip/tree/2.1.3-esp).
Note this page: https://github.com/espressif/esp-lwip/wiki/Espressif-patches-in-2.1.3-esp-on-top-of-2.1.3-rel that lists all Espressif patches (on top of `STABLE 2_1_3_RELEASE`) with links to commits and upstream patches/bugs.

closed 2022-08-22 by david-cermak