espressif/esp-lwip #40: Does LWIP support DNS over TLS

shun-dgut, 2022-02-10:


david-cermak, 2022-02-15:
Hi @shun-dgut 

You mean DNSSEC as defined in [rfc4033](https://datatracker.ietf.org/doc/html/rfc4033)? No, I'm not aware of any lwip implementation. Or did you mean some kind of carrying the DNS over (d?)tls?
The only option I could think of is to implement an external resolver and add it to lwip's DNS module using `LWIP_HOOK`s.

shun-dgut, 2022-02-16:
> Hi @shun-dgut
>
> You mean DNSSEC as defined in [rfc4033](https://datatracker.ietf.org/doc/html/rfc4033)? No, I'm not aware of any lwip implementation. Or did you mean some kind of carrying the DNS [...]

yes, i mean DNSSEC as defined in rfc4033

david-cermak, 2022-02-18:
> i means rfc7858

Thanks. This could be perhaps a little easier to implement, but unfortunately, the same story, not supported.

closed 2022-02-23 by shun-dgut