espressif/esp-lwip #41: Upstream changes/patches please

vulptex, 2022-02-21:
Please upstream generic patches to lwip please.

There are various patches more people can profit from, for me personally it is:

- DHCP option 61: Client Identifier  
- DHCP option 60: Vendor Class Identifier
- DHCP option hostname on discovery request

Why does this fork exist?

vulptex, 2022-02-21:
I already started "moving" your changes ;-)

https://github.com/lwip-tcpip/lwip/pull/10

david-cermak, 2022-03-10:
Hi @vulptex 

We plan to upstream the changes where applicable. Some of the patches need some rework, others could be simply moved to esp port layers. We're working on that and the plan is to consolidate and clean-up the patches in `2.1.3-esp` (to be added from IDF `v5.0`)

> Why does this fork exist?

Good question, ideally it shouldn't exist after the cleanup, but some changes may still not be applicable, not generic enough to be accepted to upstream (e.g. on-demand timers).
I've created this draft document listing all the esp-patches: https://github.com/espressif/esp-lwip/wiki/Espressif-patches-in-2.1.2-esp-on-top-of-2.1.2-rel and will keep it updated with the actual status and links if upstreamed/accepted.

> I already started "moving" your changes ;-)
> https://github.com/lwip-tcpip/lwip/pull/10

Oh, thanks, didn't know they accept GitHub PRs, too. I've always posted patches via https://savannah.nongnu.org/patch/?group=lwip. Also, I think would be nice to keep the original commit author when ` "moving" *your* changes ;-)`

vulptex, 2022-03-28:
i closed the PR @ lwip..will re-open when i have more time to play the "PR review" game ;-)

closed 2022-03-28 by vulptex