espressif/esp-lwip #48: IP_PORTMAP_MAX and IP_NAPT_TIMEOUT_MS_TCP configuration written to kconfig

tswen, 2022-10-13:
Hi @david-cermak 

When using the sta_ap_nat solution, if the mobile phone is connected to the AP and frequently switches the live room, it will occupy a large amount of portmaps in a short time. However, because the timeout_ms_tcp has 30 minutes, if the free_napt is not in time, the portmaps will be occupied so quickly that the mobile phone cannot access the network.

Can you write the two settings of **IP_PORTMAP_MAX** (32-255) and **IP_NAPT_TIMEOUT_MS_TCP** (10min-30min) in Kconfig, so that they can be configured according to different application scenarios.

david-cermak, 2022-10-21:
Hi @tswen 

It is possible to customize `IP_PORTMAP_MAX` even without the related Kconfig option. It's defined as

https://github.com/espressif/esp-lwip/blob/c950063f73ae11dd304b672b26aadb8d3c37296d/src/include/lwip/lwip_napt.h#L54-L56

You can set target definition to `lwip` component, just  adding these two lines in you project Makefile (place after `project()`):
```cmake
idf_component_get_property(lwip lwip COMPONENT_LIB)
target_compile_definitions(${lwip} PRIVATE "IP_PORTMAP_MAX=128")
```

On `2.1.2-esp` it's possible to update the timeout options, too:

It's already done on the `2.1.2-esp`:

https://github.com/espressif/esp-lwip/blob/c950063f73ae11dd304b672b26aadb8d3c37296d/src/include/lwip/lwip_napt.h#L62-L64

That hasn't been forward-ported to `2.1.3-esp` for some reason, sorry for that. Will fix.

tswen, 2022-10-23:
Ok, thanks for your suggestion!

closed 2022-10-23 by tswen