espressif/esp-lwip #50: gethostbyname returns invalid value

PaulMartinsen, 2022-11-16:
When I call `gethostbyname` on an IPv4 network for a device that's present on the network the `h_length` field is populated with 24 but it should be 4 (since an IPv4 address is 4 bytes long). 

For example:
```const char* pchHost = "google.com";
hostent* pHost = gethostbyname(pchHost);
  if (pHost != nullptr)
  {
    cout << pHost->h_name << ", length: " << pHost->h_length << ", type: " << pHost->h_addrtype << endl;
    if (pHost->h_addrtype == AF_INET)
    {
      int i = 0;
      while (pHost->h_addr_list[i] != 0)
      {
        struct in_addr addr;
        addr.s_addr = *(uint32_t*)pHost->h_addr_list[i];
        cout << i << ':' << inet_ntoa(addr) << endl;
        ++i;
      }
    }
  }
  else
  {
    cout << "Not found\n";
  }
```

Produces the output:
google.com, length: 24, type: 2
0:172.217.24.46

PaulMartinsen, 2022-11-16:
For reference, I'm using version 2.0.5 of the Arduino core, which is based on ESP-IDF 4.4.2.

david-cermak, 2023-03-22:
Thanks for the report! fixed in aee6b3ed0cfe67e1095d888f55b91068bdb356e4

closed 2023-03-22 by david-cermak