espressif/esp-lwip #59: Is it possible to make the SoftAP IP range have same IP with the IP that Wi-Fi STA connected to with NAT?

anhnt-dev, 2023-09-09:
Hi,
as the subject question, Is it possible? Currently, SoftAP will start a DHCP server to provide IP to the client that is connected to, and STA will get IP from Router. It will make two different networks. The SoftAP's client can ping to any device on the Router's network, but devices from the Router's network can't talk to SoftAp's client without port forwarding.

espressif-abhikroy, 2023-10-03:
Hi,
Imagine you have a router (R), an ESP32 with Wi-Fi capabilities operating as both an access point and a station (AP_STA), and another Wi-Fi station (STA1) connected to R, along with one more station (STA2) connected to AP_STA.

You want all of them to use the same network address. STA1 and STA2 can access the internet through R, but they can't communicate directly with each other as if they were on the same local network. To make that possible, you need a bridge.

Could you please share more details about your specific requirements so that I can offer a suitable solution?

david-cermak, 2024-01-23:
feel free to reopen with more details/comments.