espressif/esp-lwip #60: ARP checking not testing for it's own IP

phildimond, 2023-09-22:
https://github.com/espressif/esp-lwip/blob/3b88e84f9b05e76a1ad96e6650a795df74af23aa/src/core/ipv4/dhcp.c#L976

Device responds to it's own ARP check, detects itself and then rejects the ip address offered for DHCP. The device should check that the respondent's MAC is not it's own or (better) disable ARP responses before the check and re-enable afterward.

espressif-abhikroy, 2023-10-18:
If the device already have an IP address, the ip4_addr_cmp() function should not be executed due to the preceding check:

```C
  /* Is a DHCP client performing an ARP check? */
  if ((dhcp != NULL) && (dhcp->state == DHCP_STATE_CHECKING)) {
```
In this scenario, any ARP reply should be discarded.

If you have a different observation, kindly provide details about your network setup and all relevant logs.

david-cermak, 2024-01-23:
feel free to reopen with more details/comments.

closed 2024-01-23 by david-cermak