espressif/esp-lwip #61: Upgrade to latest lwip release

AxelLin, 2023-10-13:
FYI,
lwIP 2.2.0 released
https://lists.nongnu.org/archive/html/lwip-devel/2023-09/msg00009.html

I think esp-lwip should update to latest lwip release.

BTW, there are some bugfix in upstream but not in esp-lwip,
maybe someone should check upstream fixes.
e.g. http://git.savannah.gnu.org/cgit/lwip.git/commit/?id=574bd6e5aa57b971db7b347b8ad761342be9a02c

AxelLin, 2024-04-23:
No progress in the past 6 Months?

david-cermak, 2024-04-23:
I'm sorry it took so long, but it's already in progress from the last week.
I'll be again posting the status update of Espressif patches on top of this branch in this repo's Wiki, so you can watch our progress in https://github.com/espressif/esp-lwip/wiki/Espressif-patches-in-2.2.0-esp-on-top-of-2.2.0-rel

about the commit you mentioned, I'm not confident enough to pick it to stable branches (we'd have a similar fix in 9e62afffb49438e0b373ac3b7a831bb1bc8c3cd2 fixing a different scenario, but there might be a relation?)

david-cermak, 2024-12-16:
I'm sorry, forgot to close this issue. Branch `2.2.0-esp` was deployed in Oct-2024 as https://github.com/espressif/esp-lwip/commits/2.2.0-esp/ and used in IDF v5.4 and IDF v5.5.

closed 2024-12-16 by david-cermak

AxelLin, 2025-02-06:
@david-cermak 

FYI, lwIP 2.2.1 released
https://lists.nongnu.org/archive/html/lwip-users/2025-02/msg00001.html