espressif/esp-lwip #64: Use only IPV6

ShubhadaHarkut-eaton, 2024-02-13:
Disabling IVP4 and building the idf ethernet example code only with IPV6 support enabled leading to many compilation error 
idf version used : 4.4
os : windows 
ESP : ESP32-WROVER-B

disabling LWIP_IPV4 in opts.h and enabling LWIP_IPV6 in the same file along with other menuconfig configurations for rnabling IPV6

espressif-abhikroy, 2024-02-20:
Hi @ShubhadaHarkut-eaton,
The option to disable IPv4 is not available for IDF release v4.4 and v5.0.
It's available from IDF release v5.1 onwards.

ShubhadaHarkut-eaton, 2024-02-21:
Hi @espressif-abhikroy 
Any idea if any one has tried esp32 with ipv6 configuration as wireguard client ?

sgryphon, 2024-03-16:
I am not sure if it is relevant for your scenario, but I recently put up a pull request to update the protocol examples to work with all/any network configuration. The branch is here: https://github.com/espressif/esp-idf/pull/13250

The update examples work out of the box (no config or code changes) with all network types: IPv4-only, IPv6-only, and dual stack. (You do need to provide config for things like your WiFi password)

Note that this is not the same as turning off IPv4 -- IPv4 is still turned on, just not used. The updated example works in an IPv6-only network, if that is your objective. See the example outputs for IPv6-only network. If you have DNS64/NAT64 available, then it can even connect from an IPv6-only network to an IPv4-only server.

However, IPv4 is still actually on.

If you know IPv4 is not necessary, e.g. if you are developing a Matter device where only IPv6 is required, and want to reduce memory size and development/testing effort, then you could also turn off IPv4 completely. I also tested this scenario (current LWIP/IDF) and found a small compilation bug in the PPP code used in the examples -- this is also fixed in my PR branch, i.e. it does compile with IPv4 turn off (although there is no need to, as the examples works fine in an IPv6-only network without config changes).

david-cermak, 2024-04-19:
Closing as `won't do`, as IPv4-only is supported from `v5.1`

closed 2024-04-19 by david-cermak