espressif/esp-lwip #68: ip_addr.h:155:55: warning: the address of 'addr' will never be NULL [-Waddress]

xgnata, 2024-04-11:
Hi,

I'm not sure if these warnings are valid and if we should do something about these (except disable Waddress :) )

In file included from .pio/libdeps/esp32dev/AsyncTCP-esphome/src/AsyncTCP.h:28,
                 from .pio/libdeps/esp32dev/AsyncTCP-esphome/src/AsyncTCP.cpp:24:
.pio/libdeps/esp32dev/AsyncTCP-esphome/src/AsyncTCP.cpp: In member function 'bool AsyncClient::connect(IPAddress, uint16_t)':
/home/xavier/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/ip_addr.h:155:55: warning: the address of 'addr' will never be NULL [-Waddress]
 #define ip_addr_set_ip4_u32(ipaddr, val)  do{if(ipaddr){ip4_addr_set_u32(ip_2_ip4(ipaddr), val); \
                                                       ^
.pio/libdeps/esp32dev/AsyncTCP-esphome/src/AsyncTCP.cpp:725:5: note: in expansion of macro 'ip_addr_set_ip4_u32'
     ip_addr_set_ip4_u32(&addr, ip);
     ^~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32dev/AsyncTCP-esphome/src/AsyncTCP.cpp: In member function 'void AsyncServer::begin()':
/home/xavier/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/ip_addr.h:155:55: warning: the address of 'local_addr' will never be NULL [-Waddress]
 #define ip_addr_set_ip4_u32(ipaddr, val)  do{if(ipaddr){ip4_addr_set_u32(ip_2_ip4(ipaddr), val); \
                                                       ^
.pio/libdeps/esp32dev/AsyncTCP-esphome/src/AsyncTCP.cpp:1386:5: note: in expansion of macro 'ip_addr_set_ip4_u32'
     ip_addr_set_ip4_u32(&local_addr, _addr);

david-cermak, 2024-04-12:
Hi @xgnata 

> I'm not sure if these warnings are valid 

Yes, the warnings are perfectly valid, they tell us that the address supplied to `ip_addr_set_ip4_u32()` macro cannot be null, since we're probably taking an address of *some local variable*:
```cpp
ip_addr_set_ip4_u32(&local_addr, _addr);
```

>  if we should do something about these (except disable Waddress :) )

Yes, you can use `ip_addr_set_ip4_u32_val()` macro instead, which takes the actual address, not a pointer:
```cpp
ip_addr_set_ip4_u32_val(local_addr, _addr);
```

david-cermak, 2024-05-29:
Closing as those warnings should be handled on *AsyncTCP* layer.

closed 2024-05-29 by david-cermak