espressif/esp-lwip #70: int lwip_getaddrinfo() error

greentd, 2024-06-24:
The function lwip_getaddrinfo(const char *nodename, const char *servname,  const struct addrinfo *hints, struct addrinfo **res)  
@https://github.com/espressif/esp-lwip/blob/4297782bf9e614be1bb8605f30d46af8697fab17/src/api/netdb.c 
is  wrong,   

It causes the following test routine to fail to run
 
 https://github.com/espressif/esp-modbus/tree/master/test/tcp/mb_tcp_slave

espressif-abhikroy, 2024-07-02:
@greentd Thanks for reporting this issue. We are fixing this issue and the fix will be available soon.

AxelLin, 2024-08-06:
> @greentd Thanks for reporting this issue. We are fixing this issue and the fix will be available soon.

@espressif-abhikroy How is the status now?

espressif-abhikroy, 2024-08-06:
> > @greentd Thanks for reporting this issue. We are fixing this issue and the fix will be available soon.
> 
> @espressif-abhikroy How is the status now?

The issue was with handling of IPADDR_ANY or NULL in netdb api's. This bug was introduced with the code changes for the "multiple ip support in dns" feature(f174681317a00d209e374c66d9691ef79b4195de).
 The fix is under review and will be merged soon.

closed 2025-01-16 by Indastri